vike256/Unibot #1: py error?

Soduem, 2023-10-04:
Im assuming that im missing something for py https://cdn.discordapp.com/attachments/1112612408414961666/1159248906379599994/image.png?ex=653055ad&is=651de0ad&hm=7bc7fffcda5da42d5a1511cd1649c6dfbc8aff10d5c1f12421663f69b0275d24&

flashing error.
a more thurough step by step would help 😄

vike256, 2023-10-04:
Did you do any modifications the .ino file before uploading? And are you sure you completed these https://github.com/vike256/Unibot#raspberry-pi-pico-w-setup-for-arduino-ide

vike256, 2023-10-04:
This might help: https://stackoverflow.com/questions/54087049/fatal-python-error-initfsencoding-unable-to-load-the-file-system-codec

Soduem, 2023-10-04:
> Did you do any modifications the .ino file before uploading? And are you sure you completed these https://github.com/vike256/Unibot#raspberry-pi-pico-w-setup-for-arduino-ide

i changed the ssid and password thats about it, looking at the stackoverflow now.

Soduem, 2023-10-04:
Same exact error, even redid the environment stuff

Soduem, 2023-10-05:
> This might help: https://stackoverflow.com/questions/54087049/fatal-python-error-initfsencoding-unable-to-load-the-file-system-codec
am i missing something?? py version?? followed all steps aswell also using pico w

vike256, 2023-10-05:
I haven't had any problems uploading the .ino to the board. Have you uploaded working Arduino code to your Pico W before? What versions are you using for Python and Arduino IDE?

Soduem, 2023-10-05:
> I haven't had any problems uploading the .ino to the board. Have you uploaded working Arduino code to your Pico W before? What versions are you using for Python and Arduino IDE?

py 3.10.5 arduino 2.2.1

Soduem, 2023-10-05:
and yes i have used software on it before

Greetsew, 2023-10-05:
what version Arduino are you using @vike256

Greetsew, 2023-10-05:
can you create a short video of you uploading it and all that so we can copy your exact steps?

vike256, 2023-10-05:
> what version Arduino are you using @vike256

I am using the version 2.2.1 (latest).

> can you create a short video of you uploading it and all that so we can copy your exact steps?

Probably will someday but I don't have time right now. From what I saw in your issue #3 (Board not connected), your problem could be that you haven't selected the right com port. Go to _Arduino IDE > Tools > Port_ and try a different port.

Greetsew, 2023-10-05:
[image]
there are no ports for me

Greetsew, 2023-10-05:
i think i got it, was a faulty cable

vike256, 2023-10-05:
> i think i got it, was a faulty cable. so now i have com3 port but, still the normal error (Multiple libraries were found for "Mouse.h")

That's not a problem if it does not say error. It just selects one of the libraries automatically. The red text can be quite misleading in Arduino IDE.

Greetsew, 2023-10-05:
so that means it worked?

Greetsew, 2023-10-05:
oh i did thank you very much. i can make a video for your yt if you want?

vike256, 2023-10-05:
> oh i did thank you very much. i can make a video for your yt if you want?

Feel free to do so. You can link it to this issue.

Greetsew, 2023-10-05:
ok so it registers the colors but doesnt actully do the mouse mouvement

vike256, 2023-10-05:
> ok so it registers the colours but doesn't actually do the mouse movement

Hey, I'm closing this issue. Can you open another one with more details discussing your current issue?

closed 2023-10-05 by vike256