vike256/Unibot #2: Setup

Greetsew, 2023-10-04:
I did everything the setup says but its not running. im using raspberry pi pico w.
[image]
[image]
wifi login is also filled out

vike256, 2023-10-05:
Upload it through the Arduino IDE

Greetsew, 2023-10-05:
[image]

this is what i get from verifying and and uploading 
[image]
this is what i get when debuging

Greetsew, 2023-10-05:
ok so i didnt see the new libs u added then when i ran the bat i got this 
[image]

vike256, 2023-10-05:
Is your pc and pico connected to the same network? Also try going to your network adapter settings in Windows and enabling sharing from adapter properties.

Greetsew, 2023-10-05:
can you give me the steps to connect so ik i did it right

Greetsew, 2023-10-05:
can you show me how to find picos ip?

Greetsew, 2023-10-05:
can we talk on discord abt it so its easier ill send you a email of my discord

vike256, 2023-10-05:
Any issue is good to solve on GitHub so other users can find the solutions.

Greetsew, 2023-10-05:
can you show me how to get the ip of the pico

vike256, 2023-10-05:
Go to your Wi-Fi hotspot/router settings and you should see a list of devices and ther IPs

Greetsew, 2023-10-05:
[image]
Here?

vike256, 2023-10-05:
Try using 2.4 GHz if the Pico is not able to connect. That's what I am using.

Greetsew, 2023-10-05:
ok ill try real quick

Greetsew, 2023-10-05:
ye i got the right ip

closed 2023-10-05 by vike256

Greetsew, 2023-10-05:
but it still wont connect