vike256/Unibot #4: Mouse movement issue w/ raspberry pi pico w

Greetsew, 2023-10-05:
[image]
Registers the Colour on screen but doesn't actually do any mouse movements.

vike256, 2023-10-05:
The output doesn't seem to be showing a response from the Raspberry and it should not start another click without receiving a response. Can you check that the communication type in config.ini is set to socket?

Also this is output for the click function, is this a problem with the triggerbot or the mouse movement?

Greetsew, 2023-10-05:
"Also this is output for the click function, is this a problem with the triggerbot or the mouse movement?"
both 
[image]

vike256, 2023-10-05:
Are all the files updated to the latest version (1.2.1)?

Greetsew, 2023-10-05:
yes

vike256, 2023-10-05:
There might be a problem with cfg.client being initialized but not as an actual connection. Are you sure the Raspberry is connected properly and is receiving inputs? Using Serial.println in wifi.ino to print the command might help with debugging.

Greetsew, 2023-10-05:
can you step me through the process

vike256, 2023-10-05:
```
#include <WiFi.h>
#include <Mouse.h>

const char* ssid = "WIFI_NAME";
const char* password = "WIFI_PASSWORD";

int port = 50123;
int x = 0;
int y = 0;
int maxValue = 127;
int minValue = -127;
String cmd = "";

WiFiServer server(port);
WiFiClient client;

void setup() {
  Mouse.begin();
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }

  server.begin();
  Serial.begin(115200);
}

void loop() {
  if (!client || !client.connected()) {
    client = server.available();
  } 

  while (client.connected()) {
    if (!client.connected()) return;

    String cmd = client.readStringUntil('\r');

    if (cmd.length() > 0) {
      if (cmd[0] == 'M') {
        int commaIndex = cmd.indexOf(',');
        if (commaIndex != -1) {
          x = cmd.substring(1, commaIndex).toInt();
          y = cmd.substring(commaIndex + 1).toInt();

          if (x > maxValue) x = maxValue;
          if (x < minValue) x = minValue;
          if (y > maxValue) y = maxValue;
          if (y < minValue) y = minValue;

          Mouse.move(x, y);
        }
      } else if (cmd[0] == 'C') {
        int randomDelay = random(40, 80);
        Mouse.press(MOUSE_LEFT);
        delay(randomDelay);
        Mouse.release(MOUSE_LEFT);
      } else if (cmd[0] == 'B') {
        if (cmd[1] == '1') {
          Mouse.press(MOUSE_LEFT);
        } else if (cmd[1] == '0') {
          Mouse.release(MOUSE_LEFT);
        }
      }
      Serial.println(cmd);
      cmd = "";

      client.print("a\r");
      client.flush();
    }
    delay(1);
  }

  delay(1);
}
```

This should work. Didn't test the code but after uploading go to Serial Monitor in Arduino IDE and select 115200

Greetsew, 2023-10-05:
ok so if i put socket for communication type it doesn't connect but if I put it for none it does.

vike256, 2023-10-06:
> ok so if i put socket for communication type it doesn't connect but if I put it for none it does.

`none` doesn't try to connect to any hardware and uses ctypes for mouse input

closed 2023-10-10 by vike256