vike256/Unibot #5: Mouse movement issue with ctypes (no Arduino hardware or Raspberry)

krdeft18, 2023-10-05:
[image]
I'm trying to use this on my computer, without any additional hardware
I installed the requirements, modified the config.ini files, including:
- color lower and upper bound for Valorant's purple outline (which is [130, 60, 200] for lower and [150, 255, 255] for upper)
- monitor's refresh rate (from 165 to 144 to match my monitor)
I've tested the game on fullscreen, windowed fullscreen with raw buffer turned off. 

I managed to get it work and the terminal outputs the move value for the mouse, but ctypes doesn't execute the movement command. I think the error might be in the main function, since I don't see anything related to com_type none. 
Correct me if I'm wrong, any help is deeply appreciated !

vike256, 2023-10-05:
Try another game to test if it's working. I think Vanguard blocks WinApi mouse event calls.

closed 2023-10-10 by vike256

Le1a, 2024-05-24:
> ![movement value](https://private-user-images.githubusercontent.com/92785219/272930650-5b0d792a-2b11-46ed-b58a-543d02cdff57.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1MjAzNDEsIm5iZiI6MTcxNjUyMDA0MSwicGF0aCI6Ii85Mjc4NTIxOS8yNzI5MzA2NTAtNWIwZDc5MmEtMmIxMS00NmVkLWI1OGEtNTQzZDAyY2RmZjU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI0VDAzMDcyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzYmE5MWRjZTMyMDk2NGU1OTNhNDg4YjJjNjE2M2Y5MDk3Y2I5ZjU5YjE3NzQxYzY1NDE4ZWNiZmVhOGY3MzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.M-0J8xwrIaOMYEDW9ZDJ_EJd1jY_30XXWGI-PpkYBV4) I'm trying to use this on my computer, without any additional hardware I installed the requirements, modified the config.ini files, including:
> 
> * color lower and upper bound for Valorant's purple outline (which is [130, 60, 200] for lower and [150, 255, 255] for upper)
> * monitor's refresh rate (from 165 to 144 to match my monitor)
>   I've tested the game on fullscreen, windowed fullscreen with raw buffer turned off.
> 
> I managed to get it work and the terminal outputs the move value for the mouse, but ctypes doesn't execute the movement command. I think the error might be in the main function, since I don't see anything related to com_type none. Correct me if I'm wrong, any help is deeply appreciated !

May I ask if you have succeeded now? Can you share the config