vike256/Unibot #6: Error when trying to use a feature

wpngy, 2023-10-05:
Connected
AIM: True
SENT: Move(-127.0, 127.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)SENT: Move(-63.0, -41.0)Closed
Traceback (most recent call last):
  File "C:\Users\spngy\Downloads\Unibot-1.1_[unknowncheats.me]_\Unibot-1.1\src\main.py", line 81, in <module>
    main()
  File "C:\Users\spngy\Downloads\Unibot-1.1_[unknowncheats.me]_\Unibot-1.1\src\main.py", line 65, in main
    mouse.move(x, y)
  File "C:\Users\spngy\Downloads\Unibot-1.1_[unknowncheats.me]_\Unibot-1.1\src\mouse.py", line 21, in move
    cfg.client.sendall(command.encode())
ConnectionAbortedError: [WinError 10053] An established connection was aborted by the software in your host machine

vike256, 2023-10-05:
Give more information:
- What are you trying to do
- Config settings

The output seems like it's for `type = none` but you are getting a connection error.
Also it seems like you are using an older version, try: https://github.com/vike256/Unibot/releases/latest

wpngy, 2023-10-05:
I see, another problem im having is getting board's ip, it doesnt seem to connect to hotspot.

vike256, 2023-10-05:
> I see, another problem im having is getting board's ip, it doesnt seem to connect to hotspot.

Check `Unibot/guides/`

wpngy, 2023-10-05:
I ended up using wireless network watcher, thanks tho

closed 2023-10-05 by wpngy