vike256/Unibot #7: Fatal Python error: initfsencoding: unable to load the file system codec

Soduem, 2023-10-05:
As I already made the same issue yesterday I assume you know the error? https://cdn.discordapp.com/attachments/1112612408414961666/1159248906379599994/image.png?ex=653055ad&is=651de0ad&hm=7bc7fffcda5da42d5a1511cd1649c6dfbc8aff10d5c1f12421663f69b0275d24&

com3 port everything connected no problems connecting not a fault cable aswell.

my last issue was closed while I was sleeping

vike256, 2023-10-05:
Can't really help you since I haven't had any problems with uploading. My google searches for this issue don't bring up related information. I'm going to leave the issue open.

Soduem, 2023-10-05:
so i think i have it fixed? but which one do i select to be sure i did 
[image]

vike256, 2023-10-05:
Select the board you are using. Pico and Pico W are different products.

closed 2023-10-05 by Soduem