vike256/Unibot #17: Aiming between 2 enemies

RAPatekgenah, 2023-10-16:
Hello. All working fine. But i got problem. Aiming between 2 enemies in the center oh both head. Idk whats wrong. Color filitering issue, basically the contour merges when the 2 players are close enough

vike256, 2023-10-16:
Hey, I can't update the code yet because I am busy for now. Can you try implementing this in 'screen.py' and see of it works?

```
        if len(contours) != 0: 
             M = cv2.moments(thresh) 
             cX = int(M["m10"] / M["m00"]) 
             cY = int(M["m01"] / M["m00"]) 
             target = (cX, cY + cfg.recoil_offset)
```

RAPatekgenah, 2023-10-17:
like this?

if len(contours) != 0:
            M = cv2.moments(thresh)
            for contour in contours:
                M = cv2.moments(contour)
                cX = int(M["m10"] / M["m00"])
                cY = int(M["m01"] / M["m00"])
                distance = np.sqrt((cX - cfg.center[0])**2 + (cY - cfg.center[1] - cfg.recoil_offset)**2)
                if distance < min_distance:
                    min_distance = distance
                    target = (cX, cY + cfg.recoil_offset)

newbiecoder1714, 2023-10-17:
be my guess :)
if len(contours) != 0: 
     M = cv2.moments(thresh) 
     cX = int(M["m10"] / M["m00"]) 
     cY = int(M["m01"] / M["m00"]) 
     target = (cX, cY + cfg.recoil_offset)
return target

dont need  this
distance = np.sqrt((cX - cfg.center[0])**2 + (cY - cfg.center[1] - cfg.recoil_offset)**2)
                if distance < min_distance:

RAPatekgenah, 2023-10-17:
`from turtle import distance
import cv2
import numpy as np

import cfg

thresh = None

def get_target():
    global thresh
    target = None
    min_distance = float('inf')
    img = cfg.cam.grab(region=(cfg.region_left, int(cfg.region_top - cfg.recoil_offset), cfg.region_right, cfg.region_bottom))
    if img is not None:
        hsv = cv2.cvtColor(img, cv2.COLOR_BGR2HSV)
        mask = cv2.inRange(hsv, cfg.lower_color, cfg.upper_color)
        kernel = np.ones((3,3), np.uint8)
        dilated = cv2.dilate(mask, kernel, iterations=5)
        thresh = cv2.threshold(dilated, 60, 255, cv2.THRESH_BINARY)[1]
        contours, _ = cv2.findContours(thresh, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_NONE)

        if len(contours) != 0:
                M = cv2.moments(thresh)
                cX = int(M["m10"] / M["m00"])
                cY = int(M["m01"] / M["m00"])
                target = (cX, cY + cfg.recoil_offset)
     
return target

def get_center():
    global thresh

    target = False
    if thresh is None:
        return target

    value = 8
    if thresh[cfg.center[0] + value, cfg.center[1]] == 255:
        if thresh[cfg.center[0] - value, cfg.center[1]] == 255:
            if thresh[cfg.center[0], cfg.center[1] - value] == 255:
                if thresh[cfg.center[0], cfg.center[1] + value] == 255:
                    target = True
    thresh = None
    return target`
    
    and when i try to launch. i got this error
    Traceback (most recent call last):
  File "C:\xx\xx\xx\xx\Unibot-1.3.1\Unibot-1.3.1\src\main.py", line 9, in <module>
    import screen
  File "C:\xx\xx\xx\xx\Unibot-1.3.1\Unibot-1.3.1\src\screen.py", line 30
    return target
    ^^^^^^^^^^^^^
SyntaxError: 'return' outside function

vike256, 2023-10-17:
> ```
> from turtle import distance
> import cv2
> import numpy as np
>
> import cfg
>
>
> thresh = None
>
>
> def get_target():
>     global thresh
>     target = None
>     min_distance = float('inf')
>     img = [...]

You indented `return target` wrong. It should be in the function.

RAPatekgenah, 2023-10-17:
[image]
i fixed return target.

But that didn't solve my aim problem
still shoot between 2 enemies. how to fix this?

https://github.com/vike256/Unibot/assets/48162247/b0f03d45-deea-4a98-ae6c-f1a405f669f2

vike256, 2023-10-17:
> [image]
> i fixed return target.
> 
> But that didn't solve my aim problem
> still shoot between 2 enemies. how to fix this?
> 
> https://github.com/vike256/Unibot/assets/48162247/b0f03d45-deea-4a98-ae6c-f1a405f669f2
> 
> 
> 

Can you try changing
`target = (cX, cY + cfg.recoil_offset)`
to
`target = (cX + 2, cY + cfg.recoil_offset)`

I can't test this rn since I am on mobile.
This aims 2 pixels off center so it should prefer the target on the right.

RAPatekgenah, 2023-10-17:
I have changed it but this time it is more aim to the right, and it is still difficult to aim if the opponents are close together. Getting my aim back to the middle between the two. uncomfortable

https://github.com/vike256/Unibot/assets/48162247/947c4de7-9174-4e8e-a12a-4f2d6071692c

newbiecoder1714, 2023-10-17:
check my branch for this problem
https://github.com/vike256/Unibot/compare/main...newbiecoder1714:Unibot:newbiecoder1714-patch-1

vike256, 2023-10-17:
> Many thanks for solutions. Thank for all your help.
> i think need to be refine but idk how
> umm.. Why shaky when aiming between enemies, but hey it works now!
> 
> https://github.com/vike256/Unibot/assets/48162247/4b9fe3f8-71d6-44e8-a8ca-9c9a253a5a70
> 
> 

Lowering fps in the config might help with the shakiness.

closed 2023-10-17 by RAPatekgenah