vike256/Unibot #20: configparser.NoSectionError: No section: 'keybinds

Allmight0000, 2023-10-19:
Traceback (most recent call last):
  File "C:\Users\Administrator\Downloads\Unibot-1.4.2\Unibot-1.4.2\src\main.py", line 6, in <module>
    import cfg
  File "C:\Users\Administrator\Downloads\Unibot-1.4.2\Unibot-1.4.2\src\cfg.py", line 254, in <module>
    read_config()
  File "C:\Users\Administrator\Downloads\Unibot-1.4.2\Unibot-1.4.2\src\cfg.py", line 204, in read_config
    key_reload_config = configFile.get('keybinds', 'key_reload_config')
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python310\lib\configparser.py", line 782, in get
    d = self._unify_values(section, vars)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python310\lib\configparser.py", line 1153, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'keybinds'

vike256, 2023-10-19:
Forgot that config.ini updates were skipped by git. I'll fix in a second

vike256, 2023-10-19:
Should work now. Download v1.4.3

closed 2023-10-19 by vike256

Allmight0000, 2023-10-19:
you have a discord man?