vike256/Unibot #21: can't connect with 1.5.0

RAPatekgenah, 2023-10-22:
ERROR: Could not connect. socket.connect() takes exactly one argument (2 given)

I've read the instructions. It works on previous versions
But this time it can't . Even I reset the .ino upload from scratch. is there something wrong

vike256, 2023-10-22:
Oops. Fixed now, download v1.5.1

RAPatekgenah, 2023-10-22:
i already donwloaded and try 1.5.1 but facing new problem. Crashing issue. already connected and on. Aim True. try in Range. And unibot crash. cant work

RAPatekgenah, 2023-10-22:
can i have your discord? thanks

vike256, 2023-10-22:
Switched to an actual IDE so I don't have to do all testing manually. I'm pretty sure I got most errors fixed now, try v1.5.2

vike256, 2023-10-22:
Updated. I figured it's useless to do releases when they are filled with bugs so from now on just clone the repo.

closed 2023-10-22 by vike256