vike256/Unibot #22: Bug

1godrrty, 2023-10-22:
It force close after open game

vike256, 2023-10-22:
I fixed some bugs, try the latest version

1godrrty, 2023-10-22:
> I fixed some bugs, try the latest version

not work it can’t open

vike256, 2023-10-22:
I can't help without proper information. Can you paste your config here? And show console output

1godrrty, 2023-10-22:
> I can't help without proper information. Can you paste your config here? And show console output

[image]

1godrrty, 2023-10-22:
> I can't help without proper information. Can you paste your config here? And show console o

1godrrty, 2023-10-22:
> > I can't help without proper information. Can you paste your config here? And show console o

https://cdn.discordapp.com/attachments/1159462534743592980/1165679257524129842/image.png?ex=6547ba68&is=65354568&hm=44c238e81bb5325065619f852babffaafcbc735b16af5ba0307a9218b1c1b34b&

1godrrty, 2023-10-22:
> I can't help without proper information. Can you paste your config here? And show console output

https://cdn.discordapp.com/attachments/1159462534743592980/1165680028445593761/image.png?ex=6547bb20&is=65354620&hm=d4ed4f0b1b773b72d011e7f92a19053361e8869c544a4a0560b215a629f201e4& 

this config

closed 2023-10-22 by 1godrrty

vike256, 2023-10-22:
Updated. Clone the repo. I figured it's useless to do releases when they are filled with bugs.