vike256/Unibot #27: How

1godrrty, 2023-10-23:
How to setup config for interception driver , I try it a lot of time But it's not working.

vike256, 2023-10-23:
What is not working? Did you correctly install the driver and Unibot's python requirements?

1godrrty, 2023-10-23:
> What is not working? Did you correctly install the driver and Unibot's python requirements?
> To fix an error we need to see the error message with enough context provided.

It not have error message , I set type to driver and run ,press f2 but it not work ,I install driver like this video https://youtu.be/oJGQbK6iYLM?si=NRMmP1QxC3H0Xrps

1godrrty, 2023-10-23:
> What is not working? Did you correctly install the driver and Unibot's python requirements?
> To fix an error we need to see the error message with enough context provided.

I tried the arduino with unibot. It worked great.

1godrrty, 2023-10-23:
> > > What is not working? Did you correctly install the driver and Unibot's python requirements?
> > > To fix an error we need to see the error message with enough context provided.
> > 
> > 
> > It not have error message , I set type to driver and run ,press f2 but it not work ,I install driver like this video https://youtu.be/oJGQbK6iYLM?si=NRMmP1QxC3H0Xrps
> 
> So both the aim and triggerbot don't work with the driver? That's a 14 minute video. Are you sure you installed the driver by downloading the latest release and running the command line installer with the argument `/install`. And after that rebooted of course.
> 
> And what game are you testing this on? Some games may block the input...

I'm sure I installed the latest driver. I test on valorant

vike256, 2023-10-23:
> I test on valorant

I'm pretty sure that might be the issue. Valorant's anti-cheat blocks various types of inputs.

closed 2023-10-23 by vike256