vike256/Unibot #29: force close

1godrrty, 2023-10-27:
it force close when i play on valorant

vike256, 2023-10-27:
No sufficient information. Could be anything.

1godrrty, 2023-10-27:
> Not sufficient information. The issue could be anything.

I usually play no trouble.just got into trouble three days ago When I played it on valorant , it suddenly shut down by it self.

vike256, 2023-10-27:
> I usually play no trouble.just got into trouble three days ago When I played it on valorant , it suddenly shut down by it self.

So it's not a recurring issue? Maybe you accidentally pressed F4 which is the exit key?

1godrrty, 2023-10-28:
> > I usually play no trouble.just got into trouble three days ago When I played it on valorant , it suddenly shut down by it self.
> 
> So it's not a recurring issue? Maybe you accidentally pressed F4 which is the exit key?

I'm pretty sure I didn't press the f4 button.

vike256, 2023-10-28:
Is this a recurring issue and if it is can you provide the console output?

1godrrty, 2023-10-28:
> Is this a recurring issue and if it is can you provide the console output?

what do you mean? i not sure

vike256, 2023-10-28:
> what do you mean? i not sure
Do you have this issue frequently or was this a one time thing?

1godrrty, 2023-10-29:
> > what do you mean? i not sure
> 
> Do you have this issue frequently or was this a one time thing?

This problem is very frequent.I try to not chang any code But it force close by itself  again.

vike256, 2023-10-29:
Can you post the error message?

1godrrty, 2023-10-30:
wait i will try your new update again.

RAPatekgenah, 2023-10-31:
It should working with lastest build. Try it

1godrrty, 2023-10-31:
> It should working with lastest build. Try it

It's usable but not as good at locking targets as the old version.

1godrrty, 2023-10-31:
> Can you post the error message?

it not have a error message.I've tried it several times. i don't need to use a lastest version because it not as good at locking targets.can you edit the lastest version to have a good aimbot like the old version?

vike256, 2023-11-01:
> it not have a error message.I've tried it several times. i don't need to use a lastest version because it not as good at locking targets.can you edit the lastest version to have a good aimbot like the old version?

What do you mean? Both target detection and aiming have remained relatively same between updates.

vike256, 2023-11-01:
Issue is invalid because the user is not using the latest version. Closing.

closed 2023-11-01 by vike256