vike256/Unibot #30: ValueError: Invalid Region: Region should be in 1536x864

Kaira3535, 2023-10-30:
Traceback (most recent call last):
  File "c:\Users\DENİZ\Desktop\Unibot\src\main.py", line 85, in <module>
    main()
  File "c:\Users\DENİZ\Desktop\Unibot\src\main.py", line 32, in main
    target, trigger = screen.get_target()
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 112, in get_target
    self.debug_display()
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 127, in debug_display
    full_img = self.screenshot(self.screen_region)
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 54, in screenshot
    image = self.cam.grab(region=region)
  File "C:\Users\DENİZ\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\dxcam\dxcam.py", line 69, in grab
    self._validate_region(region)
  File "C:\Users\DENİZ\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\dxcam\dxcam.py", line 233, in _validate_region
    raise ValueError(
ValueError: Invalid Region: Region should be in 1536x864

vike256, 2023-10-30:
Can you post your config?

Kaira3535, 2023-10-30:
> Can you post your config?
> 
> The fov area moves according to given offset and recoil offset.
> 
> Either the fov is too big or the offsets move it outside the screen area.
> 
> Try to observe the issue in debug mode.

debug mode is active and I forwarded the error message to you, can you help me on discord? Ayaz.lua you can add I added you

config.ini :

[communication]
type = socket
encrypt = false
ip = 192.168.137.152
port = 50124
com_port = COM10

[screen]
detection_type = shape
upper_color = 155,255,255
lower_color = 130,50,195
fov = 394
fps = 144

[aim]
offset = 1
smooth = 0
speed = 1
x_multiplier = 1
head_height = 0.5

[recoil]
mode = move
recoil_x = 0
recoil_y = 0
max_offset = 100
recover = 0

[key_binds]
key_reload_config = F1
key_toggle_aim = F2
key_toggle_recoil = F3
key_trigger = 6
key_exit = F4

[debug]
enabled = true
display_mode = game

Kaira3535, 2023-10-30:
no matter how much I change the fov angle, even if I make it 40, even if I make it 90, I always get the same error, this is the error message I get 

Traceback (most recent call last):
  File "c:\Users\DENİZ\Desktop\Unibot\src\main.py", line 85, in <module>
    main()
  File "c:\Users\DENİZ\Desktop\Unibot\src\main.py", line 32, in main
    target, trigger = screen.get_target()
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 112, in get_target
    self.debug_display()
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 127, in debug_display
    full_img = self.screenshot(self.screen_region)
  File "c:\Users\DENİZ\Desktop\Unibot\src\screen.py", line 54, in screenshot
    image = self.cam.grab(region=region)
  File "C:\Users\DENİZ\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\dxcam\dxcam.py", line 69, in grab
    self._validate_region(region)
  File "C:\Users\DENİZ\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\dxcam\dxcam.py", line 233, in _validate_region
    raise ValueError(
ValueError: Invalid Region: Region should be in 1536x864

vike256, 2023-10-30:
The error seems to be happening when capturing the whole screen for the debug display. Does this error happen when debug is set to false?

Kaira3535, 2023-10-30:
> The error seems to be happening when capturing the whole screen for the debug display. Does the program work normally when debug is set to false?

When I make debug false, there is no error message and the system does not work, since the system is not working, I activated debug mode to understand where the error is.

Kaira3535, 2023-10-30:
strangely I was able to run the system, but I change the config setting because it aims at the body and I start main.py again and this time it does not aim at all

RAPatekgenah, 2023-10-31:
Try to find your HSV color. Make sure press F1 till ON and press F2 till aim true. All working fine no issue with lastest build

Kaira3535, 2023-10-31:
> Try to find your HSV color. Make sure press F1 till ON and press F2 till aim true. All working fine no issue with lastest build

yes I seem to have solved the problem what are your recommended settings in valorant

vike256, 2023-10-31:
Issue seems to be solved. Closing.

closed 2023-10-31 by vike256

Kaira3535, 2023-10-31:
> Issue seems to be solved. Closing.

YES THX