vike256/Unibot #37: Even with smooth set to 0, the mouse overshoots the target before circling back, as if drawing an oval.

Kaira3535, 2023-11-07:
Hello, even when I set the smooth setting to zero, the mouse rapidly moves towards the character and slightly overshoots before settling on the target. Also, when starting to shoot while running, it seems to draw a circle, which is a significant issue. After locating the position, it would be better if the cursor could directly stop on the character's head. If you could add different calculations, I would appreciate it. For example, taking into account the direction of the character's movement so that if the character is moving to the right, the mouse can be positioned accordingly. This is because there's a possibility of missing the target with the current code.

Kaira3535, 2023-11-07:
> Hello, even when I set the smooth setting to zero, the mouse rapidly moves towards the character and slightly overshoots before settling on the target. Also, when starting to shoot while running, it [...]

But there seems to be no issue with MSS; this problem only occurs when using DXCam.

Kaira3535, 2023-11-07:
Additionally, the system you have established is truly amazing. I wanted to mention that I will have a few requests soon. Could you add a flick bot and a silent bot to your system

vike256, 2023-11-07:
> But there seems to be no issue with MSS; this problem only occurs when using DXCam.

That's probably because dxcam is fsster. Try lowering speed in config.

closed 2023-11-08 by Kaira3535