vike256/Unibot #40: Unibot continues to be fantastic, and there's something else we want

Kaira3535, 2023-11-08:
**Can you include an adjustable smart AI logic in Unibot again? For instance, if I were to give an example, for use in a game like CS2, including a model introduction, and additionally to use with CUDA YOLOv8n, config, etc. If you can do this, it would really be excellent.**

Mosyexp, 2023-11-08:
that would make unibot on spar with paid hacks or rootkit, which would be superior, would love to see it implemented!

Kaira3535, 2023-11-08:
> that would make unibot on spar with paid hacks or rootkit, which would be superior, would love to see it implemented!

I am definitely waiting for a smart ai, I hope it does smart ai, that is, a scenario made with yolo, but if it can be done without model training, I think it can give excellent results.

vike256, 2023-11-08:
This is outside the scope of Unibot.

Kaira3535, 2023-11-08:
> This is outside the scope of Unibot.
> 
> Unibot is just a colorbot.

so are you planning to share a smart ai with yolov8n we would really appreciate it because you are really doing this very well
_?

closed 2023-11-08 by vike256