vike256/Unibot #42: Aiming between 2 enemies

querleyz, 2023-11-10:
Aiming between 2 enemies in the center oh both head. Idk whats wrong. Color filitering issue.

https://streamable.com/13rysd

vike256, 2023-11-10:
Try to see in debug display with mode mask. It's probably combining the enemies into one shape.

vike256, 2023-11-10:
Don't know if that's really something that can be reasonably fixed. Using pixel detection is an option

vike256, 2023-11-10:
> > Don't know if that's really something that can be reasonably fixed. Using pixel detection is an option
> 
> In Pixel Detection, it does not align exactly to the head, sometimes to the body, sometimes to the head. Not stable. DXcam and MSS both same.

It aims at the nearest pixel that is in the target color range. Might not work very well in games where only the outlines are the target color.

Mosyexp, 2023-11-10:
the potential fix for this is decreasing head detection by adding a head size to config, that will lower the head size when bot sees it and fixes the merge

closed 2023-11-10 by querleyz