vike256/Unibot #48: what should i do for arduino

rgdfssd, 2024-02-25:
I have arduino leonardo r3 , i don't know which file should a upload to arduino through arduino ide pls tell me

Ahmo9345, 2024-04-06:
Go to src folder there will be Three folders Ethernet, Serial, Wifi. Select the one you want to use only select Ethernet and WIFI if you have respected shields for them. In these folders there will be a INO file which you need to upload to your arduino leonardo r3.

Ahmo9345, 2024-04-06:
Note: If you have Ethernet or WIFI shields you to edit the INO files of your respected method.
For example: In ethernet INO you have to edit the port if you dont have that port open in your system, In WIFI INO you need to set ssid = "WIFI_NAME", password = "WIFI_PASSWORD". Hopefully this helps you BYEE BYEE

closed 2024-04-09 by vike256