vike256/Unibot #50: How to properly enable encryption?

Ahmo9345, 2024-04-06:
How to properly enable encryption as there is no proper definition in the wiki. And can you also tell how it works for nerds like me, e.g does it encrypt the mouse movements? does it encrypt the screen capture? when enable does it effect the speed?

Ahmo9345, 2024-04-06:
Also can you tell will it work with a usb host shield?

vike256, 2024-04-09:
> How to properly enable encryption as there is no proper definition in the wiki. And can you also tell how it works for nerds like me, e.g does it encrypt the mouse movements? does it encrypt the screen capture? when enable does it effect the speed?

It masks the traffic sent to the external device. It's there to make the data look more random in case an anti-cheat logs it. But it's probably useless since any anti-cheat that would log that data would still figure out that it's data related to mouse movement. This is probably the detection vector you should least worry about.

Without encryption the mouse instructions look like `M-5,0` (just a basic instruction telling to move 5 dots to the left)
With encryption the data looks like `MUYNB` (a lot less readable than the unmasked data, but still the same instruction)

Enable it by setting encryption to on both in the Unibot config and in the .ino file uploaded to the board.

> Also can you tell will it work with a usb host shield?

Yes. It's designed to work with external devices through any communication method.

closed 2024-04-11 by vike256