vike256/Unibot #54: Mouse doesn't work

copyrightsclient, 2024-04-21:
The color is detected precisely, but the mouse does not move or click. I'm using raspberry pi pico w.

config file:
```
; https://github.com/vike256/Unibot
; See Unibot Wiki for reference
; Consider donating: https://github.com/vike256#donations

[communication]
type = serial
encrypt = false
ip = 0.0.0.0
port = 50124
com_port = COM4

[screen]
detection_threshold = 3, 3
upper_color = 155, 255, 255
lower_color = 130, 50, 195
fov_x = 256
fov_y = 256
aim_fov_x = 500
aim_fov_y = 500
fps = 60
capture_method = dxcam
auto_detect_resolution = true
resolution_x = 1920
resolution_y = 1080

[aim]
offset = 0
smooth = 0.0
speed = 1.0
y_speed = 1.0
aim_height = 0.5

[recoil]
mode = move
recoil_x = 0.0
recoil_y = 0.0
max_offset = 100
recover = 0.0

[trigger]
trigger_delay = 0
trigger_randomization = 30
trigger_threshold = 8

[rapid_fire]
target_cps = 10

[key_binds]
; https://learn.microsoft.com/en-us/windows/win32/inputdev/virtual-key-codes
key_reload_config = 0x70
key_toggle_aim = 0x71
key_toggle_recoil = 0x72
key_exit = 0x73
key_trigger = 0x06
key_rapid_fire = 0x05
aim_keys = 0x01, 0x02

[debug]
enabled = true
always_on = true
display_mode = game
```

[image]

vike256, 2024-05-02:
Will not offer any help with games that have anti-cheat.

vike256, 2024-05-02:
Read the FAQ, your issue lies there. The program works. Unibot is not about avoiding anti-cheat.

copyrightsclient, 2024-05-02:
Although there is an explanation for raspberry pi pico w on the wiki page, I don't understand. The things I have done so far do not give any error while compiling the program in the normal arduino ide, I want to learn something, but I want help because it is not in an open source project similar to what you are doing.

Le1a, 2024-05-24:
> 太荒谬了，我根本无法适应像你这样的人，我的朋友@vike256

Hello, have you solved the problem? I also encountered the same problem

copyrightsclient, 2024-05-24:
> > 太荒谬了，我根本无法适应像你这样的人，我的朋友@vike256
> 
> Hello, have you solved the problem? I also encountered the same problem

No dude, I think it doesn't work because of the Valorant anti-cheat system or we made a mistake while installing it. @Le1a

Le1a, 2024-05-24:
I also found that Valorant prohibits the use of win32api, but we can use the r3 development board combined with serial communication to achieve analog mouse signals.

发自我的iPhone

------------------ Original ------------------
From: {} ***@***.***&gt;
Date: Fri,May 24,2024 5:10 PM
To: vike256/Unibot ***@***.***&gt;
Cc: Le1a ***@***.***&gt;, Mention ***@***.***&gt;
Subject: Re: [vike256/Unibot] Mouse doesn't work (Issue #54)

copyrightsclient, 2024-05-25:
I think it is a mistake to even have a game like Valorant on the computer, it is the worst game in the world.

LuckQoo, 2026-09-07:
Is it possible that you didn't press F2 and F3?