vike256/Unibot #56: Need Update (any ideas?)

RAPatekgenah, 2024-04-27:
Hello there, long time no chat Vike, I hope you are doing well.
I want to give information that Riot has been patched the Arduino things the last recent update. Vanguard blocking two same mouse input now. idk how to explain it. You know better than me.
I've been using Ethernet Shield and Arduino Leonardo, didn't work anymore, I didn't try with Host Shield maybe it can work? but recently i've been scrolling in UC found that someone who use Host Shield can't work, idk why. As far I know it has 1 input mouse but still in this case it still can't move the cursor. I don't know if this is valid or no cause i don't have it.
I want to know, how to implement HID? i think this is the only way that work? what's your thought? What requirement device I need?
Or maybe take a look by any chance, be active like gold old days if you can, i remember when you developed this masterpiece :)

RAPatekgenah, 2024-04-27:
reference to catch up: 
https://twitter.com/AntiCheatPD/status/1783326050736353535
https://www.unknowncheats.me/forum/valorant/634097-arduino-patched.html

vike256, 2024-05-02:
This is a non-issue for Unibot since it's just a hobby project and wasn't made for VALORANT specifically. The program works, I'm not here to battle against anti-cheat since I am against cheating.

closed 2024-05-02 by vike256