vike256/Unibot #58: Pi Pico W

Agis0, 2024-05-04:
Hello there,

I was searching up for more information about usage of Pi Pico W. I saw that most people use arduino for cheating. Is it working for Pi Pico W? Is it detected by Vanguard with Pi Pico W?

Sincerely,
Agis0

vike256, 2024-05-04:
Unibot works with Pico W. Avoiding anti-cheat is unrelated to this project.

closed 2024-05-04 by vike256

maropboia, 2024-05-10:
> Hello there,
> 
> I was searching up for more information about usage of Pi Pico W. I saw that most people use arduino for cheating. Is it working for Pi Pico W? Is it detected by Vanguard with Pi Pico W?
> 
> Sincerely, Agis0

I'm experimenting on bypass anti cheat with arduino but i can't find enough source code of cheats with arduino for project . Can you help by sharing what you know?