vike256/Unibot #59: Code for esp32

maropboia, 2024-05-10:
Esp32 is powerful please add support #include <BLEDevice.h>
#include <BLEHIDDevice.h>
#include <BLEMouse.h>

BLEHIDDevice* hid;
BLECharacteristic* input;

int x = 0;
int y = 0;
String cmd = "";
char symbols[] = "-,0123456789";
char code[] = "UNIBOTCYPHER";
bool encrypt = false;

void decryptCommand(String &command) {
  if (encrypt) {
    for (int i = 0; i < command.length(); i++) {
      for (int j = 0; j < sizeof(code) - 1; j++) {
        if (command[i] == code[j]) {
          command[i] = symbols[j];
          break;
        }
      }
    }
  }
}

void setup() {
  BLEDevice::init("ESP32 Mouse");
  hid = new BLEHIDDevice();
  input = hid->inputReport(1);
  hid->manufacturer()->setValue("Manufacturer");
  hid->pnp(0x02, 0xe502, 0xa111, 0x0210);
  hid->hidInfo(0x00, 0x01);
  hid->reportMap((uint8_t*)HID_REPORT_MAP, sizeof(HID_REPORT_MAP));
  hid->startServices();

  Serial.begin(115200);
  Serial.setTimeout(1);
}

void loop() {
  String cmd = Serial.readStringUntil('\r');

  if (cmd.length() > 0) {
    if (cmd[0] == 'M') {
      decryptCommand(cmd);
      int commaIndex = cmd.indexOf(',');
      if (commaIndex != -1) {
        x = cmd.substring(1, commaIndex).toInt();
        y = cmd.substring(commaIndex + 1).toInt();

        while (x != 0 || y != 0) {
          int moveX = constrain(x, -128, 127);
          int moveY = constrain(y, -128, 127);

          BLEMouse.move(moveX, moveY);

          x -= moveX;
          y -= moveY;
        }
      }
    } else if (cmd[0] == 'C') {
      int randomDelay = random(40, 80);
      BLEMouse.press(BLE_LEFT);
      delay(randomDelay);
      BLEMouse.release(BLE_LEFT);
    } else if (cmd[0] == 'B') {
      if (cmd[1] == '1') {
        BLEMouse.press(BLE_LEFT);
      } else if (cmd[1] == '0') {
        BLEMouse.release(BLE_LEFT);
      }
    }
    Serial.print("a\r\n");
    Serial.flush();
  }
}

vike256, 2024-05-10:
This project is not being worked on at the moment. Either way, GitHub issues page is not the way to contribute. You can freely fork the project and add support to your desired boards.

closed 2024-05-10 by vike256

maropboia, 2024-05-10:
I'm not sure it will work or not so i shared here. + u need to edit main code