vike256/Unibot #63: not working with rp2040

jammesop007aha, 2024-05-18:
showing no hid library. and please make a tutorial to a to z show everything

vike256, 2024-05-23:
No thanks. Ask lazy questions, get lazy answers.

closed 2024-05-23 by vike256

vike256, 2024-05-23:
All guides have been provided in the wiki. Start from there.

jaiwiwjwjwisn, 2024-05-24:
> All guides have been provided in the wiki. Start from there. As this is a cheating script in a sense, I don't find it desirable to spend my time helping people use it. I would prefer people to use this project as a learning opportunity.

> All guides have been provided in the wiki. Start from there. As this is a cheating script in a sense, I don't find it desirable to spend my time helping people use it. I would prefer people to use this project as a learning opportunity.

Brother i have one pi pico Chinese clone 16MB flash memory version, and I'm using arduino cloud (ide in internet) how to setup board manager for using unibot?