vike256/Unibot #71: issue

jammesop007aha, 2024-08-13:
Traceback (most recent call last):
  File "C:\Users\ASUS\Desktop\Unibot-main\src\main.py", line 103, in <module>
    main()
  File "C:\Users\ASUS\Desktop\Unibot-main\src\main.py", line 50, in main
    screen = Screen(config)
             ^^^^^^^^^^^^^^
  File "C:\Users\ASUS\Desktop\Unibot-main\src\screen.py", line 32, in __init__
    self.cam = dxcam.create(output_color="BGR")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\dxcam\__init__.py", line 112, in create
    return __factory.create(
           ^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\dxcam\__init__.py", line 71, in create
    camera = DXCamera(
             ^^^^^^^^^
  File "C:\Python312\Lib\site-packages\dxcam\dxcam.py", line 33, in __init__
    self._duplicator: Duplicator = Duplicator(
                                   ^^^^^^^^^^^
  File "<string>", line 6, in __init__
  File "C:\Python312\Lib\site-packages\dxcam\core\duplicator.py", line 19, in __post_init__
    output.output.DuplicateOutput(device.device, ctypes.byref(self.duplicator))
_ctypes.COMError: (-2005270524, 'The specified device interface or feature level is not supported on this system.', (None, None, None, 0, None))
Exception ignored in: <function DXCamera.__del__ at 0x00000000131B7880>
Traceback (most recent call last):
  File "C:\Python312\Lib\site-packages\dxcam\dxcam.py", line 243, in __del__
    self.release()
  File "C:\Python312\Lib\site-packages\dxcam\dxcam.py", line 238, in release
    self.stop()
  File "C:\Python312\Lib\site-packages\dxcam\dxcam.py", line 138, in stop
    if self.is_capturing:
       ^^^^^^^^^^^^^^^^^
AttributeError: 'DXCamera' object has no attribute 'is_capturing'
Exception ignored in: <function Screen.__del__ at 0x0000000013210400>
Traceback (most recent call last):
  File "C:\Users\ASUS\Desktop\Unibot-main\src\screen.py", line 83, in __del__
    del self.cam
        ^^^^^^^^
AttributeError: 'Screen' object has no attribute 'cam'

Press any key to continue ...

RAPatekgenah, 2024-08-17:
What did you do?

jammesop007aha, 2024-08-17:
> What did you do?

i  used raspberry pi pico.

vike256, 2024-10-18:
> File "C:\Python312\Lib\site-packages\dxcam\core\duplicator.py", line 19, in post_init
output.output.DuplicateOutput(device.device, ctypes.byref(self.duplicator))
_ctypes.COMError: (-2005270524, 'The specified device interface or feature level is not supported on this system.', (None, None, None, 0, None))

I'd guess it doesn't detect your screen or it's resolution. But I'll close this as I just replaced DXCam with BetterCam.

closed 2024-10-18 by vike256