vike256/Unibot #77: Mouse mouvement dont work

Nexize45, 2024-10-25:
I installed everything as instructed: color detection is working, and everything else seems to be functioning correctly. However, when I press the keys to activate the aimbot or other actions, they don't work. The only actions that respond are key_exit and key_reload_config. I suspect the issue might be related to mouse drivers. I've tried both WinAPI and Interception, but neither worked, and I prefer not to use an Arduino. I even asked ChatGPT for help, but it couldn't solve the issue.

Recap: When I run main.py, color detection works, but the inputs don't respond.

[image]
[image]
[image]

pls someone help me

vike256, 2024-10-26:
https://github.com/vike256/Unibot/wiki/FAQ-and-troubleshooting-tips#i-cant-get-it-to-work

closed 2024-10-26 by vike256

Nexize45, 2024-10-26:
i tried on aimlab too and it doesnt work

LuckQoo, 2026-09-07:
Is there any chance—is there any chance—that you didn't press F2 and F3?