vike256/Unibot #79: windows 10 problem

jammesop007aha, 2025-02-06:
Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ASUS\Downloads\Compressed\Unibot-main>pip install -r requirements.txt
Collecting bettercam==1.0.0 (from -r requirements.txt (line 1))
  Downloading bettercam-1.0.0-py3-none-any.whl.metadata (7.2 kB)
Collecting comtypes==1.4.7 (from -r requirements.txt (line 2))
  Downloading comtypes-1.4.7-py3-none-any.whl.metadata (6.5 kB)
Collecting interception-python==1.12.4 (from -r requirements.txt (line 3))
  Downloading interception_python-1.12.4-py3-none-any.whl.metadata (4.8 kB)
Collecting MouseInfo==0.1.3 (from -r requirements.txt (line 4))
  Downloading MouseInfo-0.1.3.tar.gz (10 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy==2.1.2 (from -r requirements.txt (line 5))
  Downloading numpy-2.1.2-cp313-cp313-win_amd64.whl.metadata (59 kB)
Collecting opencv-python==4.10.0.84 (from -r requirements.txt (line 6))
  Downloading opencv_python-4.10.0.84-cp37-abi3-win_amd64.whl.metadata (20 kB)
Collecting PyAutoGUI==0.9.54 (from -r requirements.txt (line 7))
  Downloading PyAutoGUI-0.9.54.tar.gz (61 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyGetWindow==0.0.9 (from -r requirements.txt (line 8))
  Downloading PyGetWindow-0.0.9.tar.gz (9.7 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyMsgBox==1.0.9 (from -r requirements.txt (line 9))
  Downloading PyMsgBox-1.0.9.tar.gz (18 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyperclip==1.9.0 (from -r requirements.txt (line 10))
  Downloading pyperclip-1.9.0.tar.gz (20 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyRect==0.2.0 (from -r requirements.txt (line 11))
  Downloading PyRect-0.2.0.tar.gz (17 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyScreeze==1.0.1 (from -r requirements.txt (line 12))
  Downloading pyscreeze-1.0.1.tar.gz (27 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyserial==3.5 (from -r requirements.txt (line 13))
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting pytweening==1.2.0 (from -r requirements.txt (line 14))
  Downloading pytweening-1.2.0.tar.gz (171 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pywin32==308 (from -r requirements.txt (line 15))
  Downloading pywin32-308-cp313-cp313-win_amd64.whl.metadata (8.3 kB)
Downloading bettercam-1.0.0-py3-none-any.whl (21 kB)
Downloading comtypes-1.4.7-py3-none-any.whl (226 kB)
Downloading interception_python-1.12.4-py3-none-any.whl (22 kB)
Downloading numpy-2.1.2-cp313-cp313-win_amd64.whl (12.6 MB)
   ---------------------------------------- 12.6/12.6 MB 2.3 MB/s eta 0:00:00
Downloading opencv_python-4.10.0.84-cp37-abi3-win_amd64.whl (38.8 MB)
   ---------------------------------------- 38.8/38.8 MB 3.5 MB/s eta 0:00:00
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Downloading pywin32-308-cp313-cp313-win_amd64.whl (6.5 MB)
   ---------------------------------------- 6.5/6.5 MB 3.8 MB/s eta 0:00:00
Building wheels for collected packages: MouseInfo, PyAutoGUI, PyGetWindow, PyMsgBox, pyperclip, PyRect, PyScreeze, pytweening
  Building wheel for MouseInfo (pyproject.toml) ... done
[... 36 lines cut ...]
C:\Users\ASUS\Downloads\Compressed\Unibot-main\src>python main.py
Traceback (most recent call last):
  File "C:\Users\ASUS\Downloads\Compressed\Unibot-main\src\main.py", line 23, in <module>
    from screen import Screen
  File "C:\Users\ASUS\Downloads\Compressed\Unibot-main\src\screen.py", line 20, in <module>
    import bettercam
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\bettercam\__init__.py", line 3, in <module>
    from bettercam.bettercam import BetterCam, Output, Device
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\bettercam\bettercam.py", line 5, in <module>
    import comtypes
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\comtypes\__init__.py", line 177, in <module>
    from comtypes._post_coinit import _shutdown
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\comtypes\_post_coinit\__init__.py", line 15, in <module>
    from comtypes._post_coinit.unknwn import _shutdown  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\comtypes\_post_coinit\unknwn.py", line 371, in <module>
    class _compointer_base(c_void_p, metaclass=_compointer_meta):
    ...<89 lines>...
            return value.QueryInterface(cls.__com_interface__)
  File "C:\Users\ASUS\AppData\Local\Programs\Python\Python313\Lib\site-packages\comtypes\_post_coinit\unknwn.py", line 93, in __new__
    p = type(_compointer_base)(
             ^^^^^^^^^^^^^^^^
NameError: name '_compointer_base' is not defined. Did you mean: '_compointer_meta'?

C:\Users\ASUS\Downloads\Compressed\Unibot-main\src>

vike256, 2025-02-06:
Seems like you're using Python version 3.13.
This is built with 3.12.7 but should work with the latest 3.12 version (3.12.9).

jammesop007aha, 2025-02-08:
> Seems like you're using Python version 3.13. This is built with 3.12.7 but should work with the latest 3.12 version (3.12.9).

so i use older python version and the issue is solved?

vike256, 2025-02-09:
> so i use older python version and the issue is solved?

Yes. It seems to be an issue with bettercam. It hasn't been updated in years.